SpikeApp/Spike #1: audio drop outs every 5 minutes

JohanDegraeve, 2018-02-19:
If it's every 5 minutes, then that's probably due to the phone muted check. This check is done very 5 minutes, although it was originally not aligned with a reading, but maybe in your version it is.

It plays a sound also, irrespective if mute check alert is on or off.

It's probably better to do the mutecheck only if the alert is on for the time.

And maybe the method to play that sound also needs to be changed. It's different than playing an alert sound
ANE, mutechecker.m, playmutesound

miguelkennedy, 2018-02-19:
I haven't aligned it with the readings... But normally the AlarmService does kick in when a new reading comes and the mutechecker fires... do you think we can play the mutechecker sound file inside the ane using the same audiosessions that we use in the spoken readings?

JohanDegraeve, 2018-02-20:
I think yes, would need to be tried

JohanDegraeve, 2018-02-20:
ok done

miguelkennedy, 2018-02-20:
Done? You changed the ane and are testing it now?

JohanDegraeve, 2018-02-20:
No not yet

miguelkennedy, 2018-02-20:
OK!

miguelkennedy, 2018-02-23:
Can we close this?

JohanDegraeve, 2018-02-23:
yes

miguelkennedy, 2018-02-23:
Done!

commit e7d97524 (2018-03-14): Popup fixes #1
closed 2018-02-23 by miguelkennedy