SpikeApp/Spike #2: Wrong message for G4, only when guide tour is selected

JohanDegraeve, 2018-02-20:
This seems only when guided tour is selected.
When selecting G4, there's a message about pairing device. But that's only applicable for blucon and G5
[image]

miguelkennedy, 2018-02-20:
So, in that case, this popu should say that the user needs to scan for the device? is that it?

JohanDegraeve, 2018-02-21:
G4, Bluereader and transmitter pl : scan for device.

G5 and Blucon : app automatically starts scanning as soon as transmitter id is set and will try to connect.
If not yet paired, the pairing request will pop up.
No need to mention that user needs to wait for pairing request, immediately start sensor.
Because maybe the iphone is already paired, usually users first try another app (dexcom app, ambrosioa app), so the device is already paired. If the user doesn't start the sensor nothing will happen.
If user user starts sensor but device not paired, he/she will get a notification anyway.

I wouldn't automatically keep the app in the foreground, because you can not know if it's needed

JohanDegraeve, 2018-02-21:
To summarize

G4, transmitter PL , and bluereader. Ask the user to scan for device, and at the same time the user can start the sensor. There's no need to do one before the other.

Blucon and G5 : explain that scanning has started, and user should now start sensor, immediately

miguelkennedy, 2018-02-21:
Deal! Will be included in the next update...

miguelkennedy, 2018-02-21:
Fixed! Out on 1.0.8...

closed 2018-02-21 by miguelkennedy