SpikeApp/Spike #5: request: Link indicator icon in status bar

moccon, 2018-02-21:
For new starters it might be unclear if the App is connected. My idea is to implement an Link icon in the status bar which indicates if the App is connected with the Transmitter.
For Blucon par example, the Ambrosia app has a link icon which switches from red to green.

miguelkennedy, 2018-02-21:
I think you have that in menu -> transmitter... We might do a statusbar icon somewhere in the future but I can't give an ETA as we have essential features to be added first..

moccon, 2018-02-21:
Ofcourse. I had some Bluetooth drops without noticing, it could be handy to always see if you're connected in the first place.  So far app is much stabler than previous versions. Very happy with it.

miguelkennedy, 2018-02-21:
Awesome! Thanks for the feedback! ;)

closed 2018-02-21 by miguelkennedy

moccon, 2018-03-06:
Hi Miguel, I still cannot tell if Spike is 'connected' and working by the look at it... Just have to wait and wait, (an icon would be so helpful) but now I found a workaround, just want to let you know!  
if I toggle the Bluetooth icon very short it (sometimes) immediately connects..