SpikeApp/Spike #6: Alarm notifications don't appear the first time

miguelkennedy, 2018-02-23:
Alarm notifications don't appear on the lock screen the first time the alarm fires. It does play the alarm sound and vibration, but the notification is not on the screen. The second time the alarm fires (and from there onwards), the notifications start to appear.

moccon, 2018-02-23:
iphone 6+ same here, would be nice to snooze it right away.

miguelkennedy, 2018-02-23:
Yes :) We're working on a fix! I've noticed that sometimes, when snoozing an alarm from the lock screen Spike crashes... This is due to a bug in the iOS notifications system... I've managed to track it and it's fixed inside Spike, avoiding crashes. The fix will be included in the next update...

JohanDegraeve, 2018-04-13:
can be closed I assume ?

miguelkennedy, 2018-04-13:
Done!

closed 2018-04-13 by miguelkennedy