SpikeApp/Spike #10: Sensor stopped if no calibration after app restart

JohanDegraeve, 2018-02-24:
- start app
- start sensor but don't calibrate yet
- close app 
- restart app

Looks as if sensor status is back to not started

Note that this is also on iosxdripreader, I'm logging it in order not to forget

miguelkennedy, 2018-02-25:
Yes, I noticed this when coding Spike... iOSxDrip does the same... Sometime ago I actually came across the code that resets the sensor status when calibrations < 2 but now I don't remember where I saw that code :D

miguelkennedy, 2018-02-25:
Maybe this is the root cause? https://github.com/SpikeApp/Spike/blob/master/src/model/ModelLocator.as#L154

JohanDegraeve, 2018-02-25:
Oh god, i don't remember why I did that.
This can be removed, in any case after a restart calibration restarts completely because the first calibration is stored in memory, not in database, until the second calibration is done.

By the way loggings in database are also not used anymore.

miguelkennedy, 2018-02-25:
Yeah, if it can be removed than do it... I+ll let you do it, I don't want to mess anything ;)

JohanDegraeve, 2018-02-25:
solved in commit f63819abbeb5bcb46527d3ada0170c12feecd5ec

closed 2018-02-25 by JohanDegraeve