SpikeApp/Spike #12: First alarm overwrites mute

JohanDegraeve, 2018-03-03:
Reported by Lukas on facebook
First planed loud high or low alarm, in my case from 23:00 on, always overwrites mute status. Once snoozed, all the alarms that follow respect the mute state of the phone…Can be very annoying, at the cinema or at a diner or something…anybody else?

JohanDegraeve, 2018-03-03:
Seems difficult to reproduce, I think I had it once when I first tried (on iosxdrip but should be the same) - after that I was not able to reproduce.

Anyone who can describe the steps to reproduce ?

JohanDegraeve, 2018-03-03:
I think I have a solution
testing

miguelkennedy, 2018-03-04:
Ok... haven't tested it yet if I can reproduce this bug... Let me know if you need me to test it.

JohanDegraeve, 2018-03-04:
I'm testing a change on iosxdripreader, I will first test a few days and probably do a TestFlight release, then port to Spike

closed 2018-03-08 by miguelkennedy