SpikeApp/Spike #13: Can’t leave Nightscout screen without restarting app

moccon, 2018-03-03:
[image]

moccon, 2018-03-03:
https://github.com/moccon/files/blob/master/ScreenRecording_03-03-2018%2013-16-49.MP4

miguelkennedy, 2018-03-04:
Does this happen always or was it a one time only glitch? What iPhone model are you using?

moccon, 2018-03-04:
Hi Miguel, iPhone 6+ and it happens every time. But i found a workaround.

Also Spike 'hangs' often, can I have a suggestion or request? 
When I glimps at Spike I see, " 03m ago" but this is sometimes actally 1hr ago ;-( 
I would prefer a timestamp with ..x.. min ago (and 'min' instead of 'm') and when 04 preferable just 4)

Just my thoughts about it, since we read from left to right, the glucose value to the left and then the timestamp in the middle (because thats were the clock is on the iPhone) then you can see with one glimps of the eye if 'now' is now, same timestamp.
Anyway just my thoughts, love the design of the new app and the customizable charts a lot.

miguelkennedy, 2018-03-04:
Hello, Jules..

That request is not possible to fulfil because Spike needs to be compatible with smaller screen iPhones like the 4S... putting "min" will make the words overlap with the glucose value on the iPhone 4s and 5S/5C.

Also, I'm sorry but I will leave the glucose on the right and timestamp on the left as it is... It's how all apps do it, including nightscout, and how most people are used to see it. Also, there are new features coming like insulin on board, carbs on board, etc... so I need to keep the layout as it is now so I can later fit the rest of the info that will come with future updates...

closed 2018-03-04 by miguelkennedy

moccon, 2018-03-04:
Hi Miguel,  too bad, i think '5 min ago'  looks more readable than  '05m ago' i always have to look twice.
Also i think such a minor change is possible. 

You said leave the timestamp as it is, but thats the problem there Is NO timestamp, '05m ago' is not a timestamp...  when your screen freezes for 20 min.. it doesn't work for me...

miguelkennedy, 2018-03-04:
As I said in my previous comment, "05m ago" doesn't work on smaller screen devices... I've already tested it some time ago...

If the screen freezes it'ss because iOS is suspending Spike... Go to settings, advanced and choose a more aggressive anti suspension option... A complete timestamp on the screen is too much data to have on the screen, it gets confusing... Have in mind that not all features are coded in Spike, the screen will be much more populated as soon as the treatments feature is added so when I do changes I need to always have in mind what's coming in the future... while the timestamp could work now, it will be to much in the future... Not worth the development time to have to take it off when treatments are added to Spike...

moccon, 2018-03-04:
Hi Miguel, ofcourse you do what you do best and have tested it, dont want you to get you extra work, just trying to think along with you. I found it clearer in the xDrip version from Johan, also it had the full! "3 Minutes ago" and it did work over there... (?)  anyway ... Just letting you know that (i think that)  _'05m ago' does not read as good as '5 min ago'._ But thats my opinion, do with it whatever you want ;) Just some user feedback! 

- Yes, Usually I have it on agressive mode.
- Just '01m ago' I wanted to check my bloodsugar, ( just '1 min ago' would be beter even here ;--) 
Spike opens, I can see the Logo, its restarting and says, "Contex3d not available!.. etc, wrong renderMode or missing device support."   It crashed again on my 6+.
Or maybe i just should buy that iPhoneX?