SpikeApp/Spike #14: dexcom follow as a alarm APP.

davidmlb, 2018-03-05:
Hi, I'm using spike with a G5 transmiter and loop, everything Works fine, but i whant to have alarms for the followers ( Android phones), I put in  setting/share/dexcom share, enabled, the username and paswords and also i send the invitation to the followers, then oi receive the invitation by e-mail and open the app, but not data is loading to the app. 

Thanks

miguelkennedy, 2018-03-05:
This is unrelated to Spike... it's probably an issue with the dexcom share servers as many users use this feature without problems...

Try this: Follower deletes you in the follower app and kills the follower app from the app switcher. You delete them as followers in Spike... Then re-add them again in Spike, follower receives invite and opens it in the follower app... This seems to work for a few selected people that have the same problem with dexcom share servers...

closed 2018-03-05 by miguelkennedy

amatsoukas, 2018-03-28:
@miguelkennedy, I am having the exact same issue, and i have tried what you propose multiple times with no success. I even tried creating a new dexcom account, and using that instead.

I have the exact same setup as @davidmlb, using a G5 transmitter with loop. Might the reason be that the transmitter is expired, and i am extending it with replaced batteries, so dexcom share is not accepting the entries because of that ?

In that case, might it be useful, to allow g5 users to upload to dexcom share with a g4 receiver serial number instead ? I noticed that if i switch to transmitter to g4 the field for the receiver serial number appears, while if i select g5 it disappears. Many of us, still own a dexcom receiver and could use that instead to upload to the dexcom servers.

miguelkennedy, 2018-03-28:
amatsoukas... please go to main menu, bug report and enable tracing for 20-25min... be sure you have dexcom share enabled during that time... then please send the bug report so I can take a look at the logs...

miguelkennedy, 2018-03-28:
Problem solved! It's working fine now...