SpikeApp/Spike #16: Spike Block with black screen

nathangiovannini3, 2018-03-10:
Spike 1.1.3 opening spike showed up a black screen also spike received readings during the block

miguelkennedy, 2018-03-10:
Hi, Nathan... This is probably an iOS thing... If you had a black screen when opening Spike it's because something in iOS prevented Spike to initiate the 3D context that allows Spike to write to the GPU card...

It's the first time this has been reported... I doubt this has something to do with Spike but I will keep an eye if someone else reports something similar...

moccon, 2018-03-11:
Hi Nathan, i had that too. Sometimes Spike opens, shows the logo and shows "Contex3d not available!.. etc, wrong renderMode or missing device support."   On my 6+. Just reboot solved it but sometimes it comes back...

nathangiovannini3, 2018-03-11:
Ok Thanks for the information

miguelkennedy, 2018-03-11:
closing this for now...

closed 2018-03-11 by miguelkennedy