SpikeApp/Spike #17: Minibug

nathangiovannini3, 2018-03-11:
Reported by: Roch Magnan In the Facebook group. 
Minibug: wrote yesterday  to say how I couldn’t get a line for a graph. Issue resolved itself after an hour or so. But it’s back this morning. One thing I’ve noticed is somehow the line tick box un checks itself with no exterior input. It looks to me now as if when you do not look at the app for hours (like when sleeping) this may occur, because I got the line back 15 mn after waking up.

miguelkennedy, 2018-03-11:
Thanks, Nathan... I'm keeping an eye on this... I've been in contact with the user...

miguelkennedy, 2018-03-15:
Closing this for now... Doesn't seem to be reporting by anyone else... Will reopen if the issue arises again...

closed 2018-03-15 by miguelkennedy