SpikeApp/Spike #18: mmol delta should be rounded to one decimal

dabear, 2018-03-13:
13:50 Glucose is 6.7 mmol
13:55 Glucose is 6.9 mmol, delta is reported as 0.17
Delta for mmol users should be rounded to the nearest decimal.

[image]

[image]

miguelkennedy, 2018-03-13:
Hi, DaBear! Is this happening in Follower or Master mode?

dabear, 2018-03-13:
Follower

hummelstrand, 2018-03-13:
I don't agree, I'd rather have better resolution with double decimal digits
 for the delta values.
-- 
     Jonas Hummelstrand
     <http://SuperFly.tv>
     +46-733-970000

miguelkennedy, 2018-03-13:
I will round it for Follower because Nightscout does round values as well so Spike is fetching rounded values... I will leave it intact for Master... Thanks for reporting this dabear...

miguelkennedy, 2018-03-13:
Fixed! Available with the next update.... Untouched for master mode because users seem to prefer having more granularity...

closed 2018-03-13 by miguelkennedy