SpikeApp/Spike #20: Overlap in scala

m-a-v, 2018-03-14:
[image]

Two values seem to overlap at the bottom.

miguelkennedy, 2018-03-14:
Hi, this is not a bug... you simply have 2 thresholds that are too close to each other... there's nothing we can do to avoid the overlap, especially because your chart is going from 2.6 to 23.4 so Spike doesn't have enough space to fit everything...

closed 2018-03-14 by miguelkennedy

m-a-v, 2018-03-14:
You could probably omti the higher values if the font will overlap. Like this the users can at least read one of the values. For me it's OK if it is not fixed.

miguelkennedy, 2018-03-15:
I'm sorry but it does not grant a fix... it's not up to spike to omit values... this same overlap also happens in Nightscout for a reason ;)