SpikeApp/Spike #24: Sometimes uploads to Nightscout missing

JohanDegraeve, 2018-04-13:
It looks as if sometimes uploads to Nightscout fail, although the logs shows the upload was successful.
I think it happens after a reconnect.

Example attached screenshot spike shows no gaps.
[image]

Here Nightscout, there's a gap at 09:15. Also in mongolab I don't find this event.
[image]

Here the logs.
The event is created at 09:15:24.610
The upload to NS shows successful at 09:15:25.482

[IssueLog.txt](https://github.com/SpikeApp/Spike/files/1906417/IssueLog.txt)

miguelkennedy, 2018-04-13:
Thank you... I noticed it too... On my TODO to check what might be wrong... The logs are helpful ;)

miguelkennedy, 2018-04-18:
Fixed... See https://github.com/SpikeApp/Spike/commit/7f0023e953c961329efc13553b63d8e61d3a6b9d

closed 2018-04-18 by miguelkennedy