SpikeApp/Spike #27: Unable to Enable Calendar for Apple Watch

AngieWing, 2018-04-23:
The "enable" and subsequent "authorize" buttons are not on my calendars screen to allow for Spike to access my calendars. I have created the new calendar in iCloud and labeled it Spike, then went into my watch app on my phone using the custom option making the Spike labeled calendar be the only calendar on my Apple Watch. I can't allow the Spike app to access the Spike calendar. Thanks in advance!

miguelkennedy, 2018-04-23:
Are you using the latest version of Spike 1.1.5? You can check which version you're using by going to Settings - About. If you're not on the latest version you can update by going to the TestFlight app...

AngieWing, 2018-04-23:
I checked and it's the most recent version: 1.1.5

Thank you for responding so quickly.

*ANGELA WING*
*630-725-8939*
*SlackerPancreas.com*
*IG: angela.wing*
angiewing0101@gmail.com

miguelkennedy, 2018-04-23:
Can you please try and reboot your phone and see if Spike is then able to access your calendars? Let me know how it goes...

AngieWing, 2018-04-23:
> If it doesn't work please post a screenshot of Spike's settings -> watch.
> Thanks!

It worked!

I will confess that it seems to have been an error on my part. However,
this may be a tad confusing for others. The part of the directions that say
'activate the "enabled" toggle at the top of this screen' seem to mean that
you are still in the watch app, not the spike app, since the watch app is
the last page the directions have you on. I didn't see the "enable" toggle
until you told me to send you a screenshot of that page in the Spike app.
Just an FYI!

   -

   *Sync calendar to the watch:* Open the Watch app on your device by
   clicking the Watch icon directly from your home screen. Go to "Calendar"
   and under the "Calendars" section click "Custom" and make sure the calendar
   you created in the previous step is selected. Try to have as less calendars
   as possible synced to your Apple Watch otherwise your glucose values might
   get overlapped by other calendar's events.
   -

   *Authorize Spike:* Activate the "Enabled" toggle at the top of this
   screen, click the "Authorize" button and allow Spike to access your
   calendars. If by any chance you deny access to your calendars and want to
   enable it at a later time you'll need to go to your device settings,
   privacy, calendars and turn on "Spike".

Thank you for the help. This app is truly amazing. I am going to be making
a donation once pay day hits my bank account. Thanks again for all you're
doing to assist this community.

*ANGELA WING*
*630-725-8939*
*SlackerPancreas.com*
*IG: angela.wing*
angiewing0101@gmail.com