SpikeApp/Spike #30: Initial calibration problem

francesc0-cgm, 2018-04-29:
Since new initial single calibration alghoritm was inserted i am having a lot of problem with accuracy in the first hours of sensor life. Seems it starts to over or underestimate bg till a new calibration is inserted. Usually i insert the sensor continuing last session, i wait half an hour so initial noise goes away and i put the measure when bgs are stable (using loop). With new alghoritm bgs seems crazy till i insert a second calibration.

JohanDegraeve, 2018-04-29:
I propose we wait till we get more feedback from other users.

miguelkennedy, 2018-04-30:
Agree... I''m not facing any issues myself...

m-a-v, 2018-05-09:
I didn’t face any problems either (Libre).

davidmlb, 2018-06-22:
no problems at all, ( libre ang G5), but I use two blood glucose strips to be sure the value I put in the calibration is a real value

brookslively, 2018-06-22:
I have this same problem. I've gone through 4 or so sensors since I've started and run into the same issue each time. I haven't figured out a good solution yet other than to wait and recalibrate after a bit. I'm using a G5 with an iPhone 6s currently running iOS 11.3.1 and on the latest version of Spike. Always use a 2+ hr soak. An example from my nightscout report is below. Let me know if I can help with testing anything out.

[image]

brookslively, 2018-06-23:
Is there data I could collect when I start a new sensor that could be me/us troubleshoot?

francesc0-cgm, 2018-06-23:
Imho some sensors have much noise than others. I found that with only calibration they started crazy then calibrating it again it went ok.

miguelkennedy, 2018-06-26:
When this happens just wait 10-15 minutes and insert a new calibration (overriding the previous one). This tends to fix the issue... It seems to be related to sensor noise...

brookslively, 2018-07-25:
I just changed my transmitter this evening and ran into the same issue. I was hoping changing the transmitter would help, but this sensor startup calibration issue happens every time for me without fail. I feel like I have to be doing something wrong...

[image]

miguelkennedy, 2018-07-25:
You added you initial calibration when your BG was going down so that confused the algorithm... when that happens just wait for a new reading an insert a new calibration...

miguelkennedy, 2019-01-22:
@JohanDegraeve found a fix for this. It will be released Wednesday (Spike version 3.5.2). Closing this issue now as it's 100% fixed. Just download the new update when we release it Wednesday.

closed 2019-01-22 by miguelkennedy