SpikeApp/Spike #32: Suggestion: use stored data to fill gaps....

NerdyEE, 2018-05-06:
When I referee soccer games I do not wear my iPhone or an iWatch and Dexcom is able to update the readings while I was away from my iPhone. If you were to use this feature, gaps would go away. As of now I can’t use Spike while I referee soccer matches. I don’t want to go back to Dexcom.  Please consider this suggestion.....

NerdyEE, 2018-05-07:
When using Dexcom, if I leave my iPhone for less than 90 minutes (walk away from phone), the data will get down loaded eventually. I swear Dexcom drops data but will recover it most times...

Let me know if you need more explanation....

miguelkennedy, 2018-05-07:
Backfill data is not possible because Spike reads raw values from the transmitter and then applies it's own calibrations... Dexcom transmitters don't send backfill data in raw format. It works for Dexcom because the Dexcom app doesn't do calibrations in-app, it sends those calibrations directly to the transmitter and gets all readings already calibrated...

We use raw data instead so Spike can work with refurbished and expired transmitters. I'm afraid it's not possible to implement your request...

closed 2018-05-07 by miguelkennedy

NerdyEE, 2018-05-07:
Hi Miguel and thanks for explaining.  Are any of the following possible?

Config Xmitter so data stored is more RAW?

Back calculate RAW data from the calibrated?

This path would make it so people can walk away from iPhone and not miss anything. Also makes it so you don’t have to have so many settings trying to close data holes....

miguelkennedy, 2018-05-07:
Not possible... you can't configure the transmitter to do that... it comes pre-configured from factory...

NerdyEE, 2018-05-07:
So back calculate RAW data from their stored data???  I’m trying to help and not be a pest....

If I use an iWatch, will that help?

I have all options enabled and I still get data holes but battery is hit heavily.....

miguelkennedy, 2018-05-07:
It's not possible to calculate raw from stored data... Dexcom transmitters don't store raw data... You're asking for something that it's impossible to achieve...

NerdyEE, 2018-05-08:
Hi Miguel,
I understand that the data is not being stored in a format that the tool can directly use. But having huge holes in data when I referee is not an option either. I really need to know what is happening when I’m on the soccer pitch or if I walk away from my phone at anytime. I’m very new to T1D....

Data can be cross correlated. You know what Dexcom is thinking and you have your calculated results from the raw data you use. You can see when you are missing data so stitch the data, color it PINK to differentiate if needed, but don’t toss data. A soccer game lasts over 90 minutes, I have warm-up and game reports before I’m back at phone. Heck even a bathroom visit or full shower will cause gaps...

Would an iWatch help me gather data while on field?

I love your tool, but this is one missing feature that the Dexcom tool has and Spike doesn’t....

Sorry to be an obvious annoying pest to you....

NerdyEE, 2018-05-08:
And you are rude and do not answer direct questions I have asked multiple times now!!! I am trying to help you by suggesting an alternative method that should be EASY to implement with what you have. So your claim that this tool does everything Dexcom does and more is total false advertising!!! The data holes really make this tool look like a joke with Band-Aid patches that hammer a battery, Apple Carplay or both....

Have an ordinary day Miguel....