SpikeApp/Spike #33: Hammers AppleCar Play

NerdyEE, 2018-05-07:
If Spike is running and plugged into AppleCar Play on my 2017 Bolt, audio will dump into Apple iPod mode and out of Sirius XM or other audio source....

Dexcom has similar issue but only if opened (top). Previously Dexcom would jump this mode every 5 minutes but would jump back to previous audio source. Spike locks in iPod mode within in seconds of plug in....

miguelkennedy, 2018-05-07:
Try going to Spike's settings, advanced and enable alternative method #1 when you're in the car... See if it fixes it... After you leave the car be sure to turn off this option, it wastes more battery

NerdyEE, 2018-05-07:
Thanks Miguel!
I think that cured it as I also had to increase the settings as the collection average was not looking good. Thus why I thought the use of previous data would help you and everyone...

So if I got an iWatch for referee work, would Spike gather data while on soccer pitch and away from iPhone??? Or still data holes....

-J

NerdyEE, 2018-05-07:
Hi again Miguel,
I left other comments on the other thread about past data...

Just tried iPhone on my car and initial plug in to car does still change mode to iPOD mode. Yes I can jump back to XM but undesired feature to change the CarPlay modes.....

miguelkennedy, 2018-05-07:
Have you tried enabling alternative method before entering the car?

NerdyEE, 2018-05-07:
Yes everything was enabled when I plugged in....

-J

miguelkennedy, 2018-05-07:
Then as a last resort try to keep Spike open and without locking the screen (click the three dots menu on the upper right corner and select no lock"... When Spike is open and the screen doesn't lock Spike will not use iPhone's audio

closed 2018-05-08 by miguelkennedy