SpikeApp/Spike #39: API Enhancement request.

PedanticAvenger, 2018-05-11:
Hi.

To more closely match nightscout endpoints, would it be possible to get a /status.json endpoing added that can return (at minimum) the following info pieces of the full nightscout response (sample pasted at bottom)?
settings:{"units":"mgdl or mmol"}
and
"thresholds":{"bgHigh":X,"bgTargetTop":Y,"bgTargetBottom":A,"bgLow":B}

{"status":"ok","name":"Nightscout","version":"0.10.2-release-20171201","serverTime":"2018-05-11T18:55:06.469Z","serverTimeEpoch":1526064906469,"apiEnabled":true,"careportalEnabled":true,"boluscalcEnabled":true,"head":"b5c6b7ce818c71cd7407adfccf28cfee3c06695f","settings":{"units":"mmol","timeFormat":12,"nightMode":false,"editMode":true,"showRawbg":"never","customTitle":"Nightscout","theme":"colors","alarmUrgentHigh":false,"alarmUrgentHighMins":[30,60,90,120],"alarmHigh":false,"alarmHighMins":[30,60,90,120],"alarmLow":false,"alarmLowMins":[15,30,45,60],"alarmUrgentLow":false,"alarmUrgentLowMins":[15,30,45],"alarmUrgentMins":[30,60,90,120],"alarmWarnMins":[30,60,90,120],"alarmTimeagoWarn":false,"alarmTimeagoWarnMins":15,"alarmTimeagoUrgent":false,"alarmTimeagoUrgentMins":30,"language":"en","scaleY":"log","showPlugins":"careportal delta direction upbat","showForecast":"ar2","focusHours":3,"heartbeat":60,"baseURL":"","authDefaultRoles":"readable","thresholds":{"bgHigh":260,"bgTargetTop":180,"bgTargetBottom":80,"bgLow":55},"DEFAULT_FEATURES":["bgnow","delta","direction","timeago","devicestatus","upbat","errorcodes","profile"],"alarmTypes":["predict"],"enable":["basal","bwp","cage","careportal","cob","iob","rawbg","sage","iage","treatmentnotify","boluscalc","profile","treatmentnotify","bgnow","delta","direction","timeago","devicestatus","upbat","errorcodes","ar2"]},"extendedSettings":{"basal":{"render":"none"},"profile":{"multiple":false},"devicestatus":{"advanced":true}},"authorized":null}

PedanticAvenger, 2018-05-13:
> Spike already has a status endpoint
>
> http://127.0.0.1:1979/api/v1/status.json
>
> But probably not all info you want is there as this endpoint was added
> some time ago and most info was not relevant to Spike at the time...
>
> I will see if I can add the missing info soon

Thanks!  I'm not an IOS user myself but the Fitbit watchface I developed is
getting a few Spike users asking for it to work.
I'll update my docs to point to the appropriate URL.

miguelkennedy, 2018-05-13:
https://github.com/PedanticAvenger/FlashCGM

This one?

miguelkennedy, 2018-05-13:
Actually... Just checked and I already implemented thresholds in Spike awhile ago

Here's a sample output from http://127.0.0.1:1979/api/v1/status.json

{
"status": "ok",
"apiEnabled": true,
"serverTimeEpoch": 1526182575945,
"name": "Nightscout",
"careportalEnabled": false,
"authorized": null,
"settings": {
"editMode": "on",
"alarmTimeagoUrgentMins": "30",
"enable": [
"careportal",
"iob",
"cob",
"bwp",
"treatmentnotify",
"basal",
"pushover",
"maker",
"sage",
"boluscalc",
"rawbg",
"upbat",
"pushover",
"treatmentnotify",
"bgnow",
"delta",
"direction",
"timeago",
"devicestatus",
"errorcodes",
"profile",
"simplealarms"
],
"DEFAULT_FEATURES": [
"bgnow",
"delta",
"direction",
"timeago",
"devicestatus",
"upbat",
"errorcodes",
"profile"
],
"timeFormat": 24,
"units": "mg/dl",
"alarmTimeagoWarnMins": 15,
"focusHours": 3,
"alarmUrgentHighMins": [
30,
45,
60,
90,
120
],
"showRawbg": "always",
"alarmHigh": true,
"language": "en",
"showPlugins": "careportal iob cob bwp treatmentnotify basal pushover maker sage boluscalc rawbg upbat delta direction upbat rawbg",
"alarmTimeagoUrgent": true,
"customTitle": "Spike",
"alarmUrgentHigh": true,
"alarmTypes": [
"simple"
],
"authDefaultRoles": "readable",
"alarmTimeagoWarn": true,
"alarmUrgentLow": true,
"thresholds": {
"bgTargetBottom": 70,
"bgLow": 55,
"bgHigh": 170,
"bgTargetTop": 200
},
"alarmUrgentMins": [
30,
60,
90,
120
],
"heartbeat": "10",
"alarmLowMins": [
15,
30,
45,
60,
90,
120
],
"theme": "colors",
"scaleY": "log-dynamic",
"nightMode": false,
"showForecast": "ar2",
"alarmUrgentLowMins": [
5,
15,
30,
45,
60,
90,
120
],
"alarmHighMins": [
30,
45,
60,
90,
120
],
"alarmWarnMins": [
30,
60,
90,
120
],
"alarmLow": true,
"baseURL": "http://127.0.0.1:1979"
},
"boluscalcEnabled": false,
"version": "0.10.3-dev-20171205",
"head": "",
"extendedSettings": {
"basal": {
"render": "default"
},
"timeago": {
"enableAlerts": true
},
"sage": {
"warn": 504,
"enableAlerts": true,
"info": 336
},
"errorcodes": {
"warn": "1 2 3 4 5 6 7 8",
"info": "1 2 3 4 5 6 7 8",
"urgent": "9 10"
}
},
"serverTime": "2018-05-13T03:36:15.000+0000"
}

As you can see, both units and thresholds are already present in Spike's internal server

PedanticAvenger, 2018-05-14:
Heh, Thanks!  I'll update my docs.

And yes, that is the watchface. Has a Bug I"m about to fix however if you
are not using mmol units.

PedanticAvenger, 2018-05-14:
Is your sgv URL  http://127.0.0.1:1979/api/v1/sgv.json
<http://127.0.0.1:1979/api/v1/status.json> ?

miguelkennedy, 2018-05-14:
Yes...

You can also use count to get the amount of readings you need... Example http://127.0.0.1:1979/api/v1/sgv.json?count=12

PedanticAvenger, 2018-05-14:
Thanks for the timely responses!  Looks like we are good here.

closed 2018-05-14 by PedanticAvenger