SpikeApp/Spike #46: Nightscout Transmission

anapaulalago, 2018-05-21:
I am experiencing more transmission issues to Nighscout than usual on 2.0.9 update (using BluCon NightRider). Everything else (calendar update to use on Watch, Badge, ...) working flawlessly when the app is running on background. As soon as I open the app data is transmitted.  Running iPhone X with just Spike on the background and Aggressive Mode. Trying to set to Very Aggressive (with some battery compromise). Anything that I am missing?

miguelkennedy, 2018-05-21:
Hi... We haven't changed anything Bluetooth related in 2.0.9... 2.0.9 update was just regarding some changes to the 3D engine... Have you tried doing a hard reboot?  https://ios.gadgethacks.com/how-to/force-restart-iphone-x-when-its-acting-up-0180016/

anapaulalago, 2018-05-21:
Hello, a soft reset was enough. Thanks.

closed 2018-05-21 by anapaulalago