SpikeApp/Spike #117: Feature Request: Basal Insulin Displayed on Chart

sooperlooper, 2018-08-07:
[image]

This feature would help Spikers understand how their basal insulin is affecting their BG and allow them to make adjustments accordingly. It's especially useful for pumpers as fine tuning basal rates is a major component of insulin pump therapy.

In addition, basal rates for DIY Loopers are automatically adjusted every 5 mins. Knowing what our basal rate is every 5 mins is just as important as knowing what our BG reading is every 5 mins. Having it graphically displayed like it is in Nightscout helps us determine if the algorithm has been working correctly (ie. setting the appropriate basal rate for our current glucose level) and if we need to adjust certain settings.  

Basal rate changes could be read directly from the Loop app or from Nightscout. 

Thank you so much for your consideration of this important feature!

analogmonster, 2018-12-14:
Agree I would like this, miss it from nightscout

daktau, 2018-12-31:
I concur, this would be an extremely useful feature to introduce.

sooperlooper, 2019-01-22:
@miguelkennedy I was just wondering if this feature is still on the todo list?

sooperlooper, 2019-01-22:
🎉🎊💯 I can't even imagine how complex this must be. It will definitely take Spike to the next level. Thank you so much for improving all of our lives.

closed 2019-03-28 by sooperlooper