SpikeApp/Spike #143: Glucose Readings are a Straight Line

rassob, 2018-11-22:
My current G5 transmitter stats are:

A: 316
B: 305
R: 627

When I restart my G5 Dexcom sensor, the readings go as a straight line as my initial calibration value. 

For instance, if my initial calibration blood glucose is 150mg/dL, then it reads as 150 +/- 2 for hours. If it is 240, then it reads 240...etc.

I don't know if this is due to the battery life or the sensor.  At some point, it recommended me to "reset the transmitter clock", which I accepted.

Thanks

awtyst, 2018-11-23:
Hello,

My doctor asking present data from my treatment, what is the best solution
to extract data from Spike app?

Thank you

prakashdevadas, 2018-12-13:
Hi,
Even I am having the same straight line with calibrated value. 
I am using Spike with BluCon Nightrider. Status of connection shows connected.
Can anyone help me to fix this issue
Thanks in advance 
Regards

miguelkennedy, 2018-12-26:
That just means your sensor has expired. When a sensor reaches 14 days and 12 hours of use it will stop updating readings and will send to Spike always the same reading. Just change your sensor and start a new one.

closed 2018-12-26 by miguelkennedy