SpikeApp/Spike #148: Connection to Transmiter

anapaulalago, 2018-12-18:
I am having a hard time to keep BluCon connected to Spike. Not an issue with the Sensor or BluCon (since I have tried with two I own). Spike has gotten less reliable (regarding keeping an active connection - every mode till Very Aggressive was tested) over time to the point where LinkBluCon is more stable. Not sure if Spike enters a search mode if no data is received... Sad since it does not alert caregivers or send to Nightscout. Also, no more updates let me wonder if Spike is still an ongoing project. I  was contributing to let the project alive :(

lrdHenke, 2018-12-25:
Having the same issue. :/

miguelkennedy, 2018-12-26:
I advise switching to MiaoMiao since it's MUCH more reliable. I personally use a MiaoMiao as my daily driver with Spike and have literally zero connection issues. BlueCon is extremely unstable when it comes to Bluetooth. We spent a lot of time trying to make Spike as reliable as possible with BluCon and this is as far as we can go, BluCon is just not stable Bluetooth wise. Some BluCon users report zero issues with Spike while others report more issues which makes us believe that some BluCon units are just more unreliable than others. Unless BluCon changes their BT protocol there's not much more we can do to make it better :(

Spike is constantly being updated. A new update was released yesterday that took more than 200h of hard work. AFAIK, Spike is and continues to be the diabetes open source project with the fastest and most regular update turnaround :)

anapaulalago, 2019-01-01:
Have switched to MiaoMiao and it is MUCH more reliable than BluCon, have been using for 5 days now and zero connection issues, even when I leave my phone for a while it automatically reconnects, something unimaginable before. Than you @miguelkennedy for the suggestion! Now I can use Spike and will keep on supporting your hard work!

lrdHenke, 2019-01-03:
> Having the same issue. :/

Actually, I noticed it is very stable for me. Detected my daughter disconnected spike on her end....

clarkritchie, 2019-01-09:
We've found that the Spike app freezes almost nightly requiring a force kill on my daughter's iPhone 5.

closed 2019-01-22 by miguelkennedy