SpikeApp/Spike #151: Enhancement: Pull Data From Loop App

sooperlooper, 2018-12-26:
Hello!

I was wondering if Spike could pull data from the Loop app instead of Nightscout? 

I've noticed that some of the data displayed in Spike is often delayed since the flow uses Nightscout as a proxy and requires an internet connection.  The same data is available offline from the Loop app so I think pulling the data directly would really improve Spike for Loop users.

This is some of the data I'm referring to:

- Predictions
- Pump Battery
- Pump Reservoir
- COB
- IOB
- Minutes since last Loop
- Current Basal

Thank you so much for your consideration.

miguelkennedy, 2018-12-26:
No... that data is not available for Spike to pull offline. Loop doesn't run an internal webserver like Spike nor does it provide an API to connect to.

Data from Loop should not be delayed by more than 15 seconds. Also, every time you open Spike or bring it to the foreground Spike automatically refreshes that data. If you press the predictions pill you'll also find a button to manually refresh that data if you so choose to. That's really the best I can do since there's no way to connect to Loop directly.

sooperlooper, 2018-12-26:
[image]
[image]

I see. Oh well, maybe that’s something Loop could support in the future. Right now I’m on vacation with intermittent wifi so that’s probably why I’m seeing the delays. Thanks so much for the response!

closed 2018-12-26 by sooperlooper