SpikeApp/Spike #154: Bolus Calculator

minimalnetwork, 2018-12-30:
Nice job! This is a great and a very complete and useful App! Congrats! There is a little bug un Bolus Calculator. Carbs field of Final Calculations, remains to 0 when Carbs Precision is configured to 0.5. You must to manually increase this field at Final Calculation.

miguelkennedy, 2019-01-22:
Hi.

Can you give a little more detail? Can you tell me exact steps to reproduce the issue? Thanks!

minimalnetwork, 2019-01-27:
Hi, you must to go to Settings > Treatments > Bolus Calculator and adjust Carbs Precision to 0.5. Then go to Bolus Calculator and you must to increase first Carbs field. If you scroll down, you will see that second Carbs field remains to 0 and you have to increase it manually.