SpikeApp/Spike #163: Dexcom g6 backfill data

analogmonster, 2019-01-29:
Hi. I tried to find this out on the wiki but I didn’t find an answer. Should I be getting backfilled readings? Right now if I miss a reading 5 minutes ago it doesn’t backfill, but back when I used the official app it would backfill the last few hours

bmammon, 2019-01-30:
[image]

xpucuto, 2019-01-30:
I just want to add that many families in my group are having problems with the gaps in the graph despite the aggressive mode and would really love spike to have that feature . I am the most excited actually because this is what makes me use the official app atm. XDrip has it for a quite some time and i think it also uses the Dexcom algorithm which I have no idea whether is possible to be easily implemented in spike .
 I will happily test it when needed ;)

JohanDegraeve, 2019-01-30:
What I suggest : I'm currently learning and writing Swift. Today I have MiaoMiao and xdrip/xbridge working.
I created some classes which allow to easily add new types of transmitters (bluetooth code in Spike has become a mess, difficult to maintain - my code by the way :) ) (the Swift code is here : https://github.com/JohanDegraeve/xdripswift/tree/master/xdrip/Transmitter)

One advantage is that I can easily copy code from other projects, for instance https://github.com/LoopKit/CGMBLEKit, which also supports G5 and G6 (with backfill), because this is written in the same language: Swift.

In Spike we use Native Extensions to call objective-c code. It should be possible to also add swift code. Which means if I could adapt the Spike Native Extension so that it can use the new classes I have in Swift, which I could also extend to support backfilling for G5 and G6, then we have Spike supporting G5 and G6 backfilling.

Not sure however if this will work (never tried objective-c using Swift in an AIR Native extensions)
and I would need help from @miguelkennedy 

And I would need a G5 or G6. Possibily I'll switch back to Dexcom in the coming weeks/months. Right now I'm using Dexcom because I was testing xdrip

bmammon, 2019-01-30:
Johan, I'm happy to donate G5 supplies for this cause. Just message me your address, I can send a sensor. I have dead transmitters also if you need, just let me know.

analogmonster, 2019-01-30:
Well that clears things up a little for me. Its not a problem at my end its just not supported yet. 

I’m going to close this as its known issue, but please feel free to continue talking through how to support development

closed 2019-01-30 by analogmonster