SpikeApp/Spike #172: Will spike support Eversense?

udayshah33, 2019-02-26:
I just got an Eversense implanted. Exploring options to run openaps through Spike for a MDT 715. Please help. Thanks

m-a-v, 2019-02-28:
Currently, Eversense is not supported.

There is an workaround for Android (see here https://github.com/NightscoutFoundation/xDrip/issues/616). Open source apps like Spike or xDrip+ ideally would need to have access to the protocol or to the transmitter/hardware to investiage it further. As far as I know, currently no Spike dev has access to a Eversense transmitter.

DagmaraEN, 2019-05-03:
Is there a way to share data to iOS eversense now from Android eversense??

timmy354321, 2019-06-22:
I am using the spike app and would like this feature since I'm considering switching to eversense!

amigooo, 2019-07-01:
I used the spike app. Now I switched from Libre to Eversense. It would be great if spike support the Eversense. Can I help in any way?

olafkerkhof, 2020-01-10:
can there please be an app for IOS use so i can use my apple watch as a stand alone collector from my eversense XL!?