SpikeApp/Spike #186: ignition app and spike app cannot be verified App no longer works

usmikedukes, 2019-04-06:
I upgraded to the New Ignition app and upgraded to new version of spike. everything worked well yesterday.  this afternoon nothing work; getting message that apps not verified. seems Apple does not like Ignition and Spike since they are not Apple.

What to do?
Appreciate the effort to keep the program alive.  Love the software and cannot live without it.

gtanse, 2019-04-06:
I had the same result from the app this morning so I had to delete it and go back to the Dexcom app.

abdulhuj, 2019-04-07:
Dear Mr.Miguel Kennedy,
could you please make available a compiled  IPA file of Spike app so we can easily code sign it with our own certificate since ignition in the past 2 days had its certificate revoked by apple twice. You will really save us from the fear and hassle of worrying that overnight apple could simply revoke ignition certificates and leave us unguarded without spike app. Many thanks

usmikedukes, 2019-04-07:
I was able to get  Ignition to load and verify this afternoon and this evening was able to get Spike to successfully load and verify.  It was definitely a issue of revoked certificate.  I hope this resolves the issue.

glberen, 2019-04-07:
Spike is working again for me, except that Apple Health doesn't see it at all, so the readings aren't being written to the health db.

dxclancy, 2019-04-07:
Redownloaded and tried at 10:30am PST. 
I was not able to get spike working. “Unable to verify app”.

leoggit, 2019-04-08:
I cannot even install the app under Ignition now. It keeps giving the error message failed please try later!

Also even though I uninstalled the previous installation, there is an empty Spike folder and an apparently corrupted Spike database in Files on iPhone, neither of which can be deleted. 

As a Miaomiao owner I have switched to the Tomato App by the same company. It is not as good as Spike, but the benefit is far outweighed by it being a proper App on the App Store.  http://tomato.cool/
It has most of the same functionality, but requires some digging around to set it up. 

Spike is now no longer worth the hassle. I hope my recent donation will be put towards getting it properly approved by Apple and then I would be delighted to switch back.

Treedate, 2019-04-10:
Same to me. It is not longer working. I did everything I  could. I uninstalled the app many times, I did the verification many times, I installed different versions, but anything has worked. What can I do?

m-a-v, 2019-04-14:
With version 3.6.2 there are problems for a lot of people. An alternative is being worked on.

closed 2019-04-14 by m-a-v