SpikeApp/Spike #187: Availability of Spike App IPA file

abdulhuj, 2019-04-07:
Dear Mr.Miguel Kennedy,
could you please make available a **compiled IPA file of Spike app** so we can easily code sign it with our own certificate since ignition in the past 2 days had its certificate revoked by apple twice. You will really save us from the fear and hassle of worrying that overnight apple could simply revoke ignition certificates and leave us unguarded without spike app. Many thanks

alevar1975, 2019-04-07:
Dear Miguel, I’ve got the same issue. Till April 6th version 3.6.2 worked fine wirh the ignition certificate. Since yesterday, the certificate gets revoked and reinstalling Spike does not solve the issue. Would you be so kind to post another solution. Thx a mill:)

caiobarreto, 2019-04-09:
The same here :(

ramzxxx, 2019-04-10:
> Dear Mr.Miguel Kennedy,
> could you please make available a **compiled IPA file of Spike app** so we can easily code sign it with our own certificate since ignition in the past 2 days had its [...]

for anyone who has problems with spike app and your using "miao miao" device with the libre cgm the "tomato" app  from official apple app store seems to work with loop app (V1.9.5) tomato sends miao miamo reading to loop every 5 mins (in loop i have set the cgm setting to the insulin pump (my pump minimed 722)

ramzxxx, 2019-04-10:
"tomato" app from the official apple app store works with "miao miao" /loop app can be used a temporary fix until problems with spike are resolved

ramzxxx, 2019-04-10:
I hope this info helps some people out (first time on here total NOOB)

miguelkennedy, 2019-04-10:
Dear users, I'm working on a final and definite solution for Spike... I've announced it here https://www.facebook.com/groups/spikeapp/permalink/2279140039073394/

I ask you all just a little more patience. Spike is coming back but I still need time to have everything ready.

Thank you!

closed 2019-04-10 by miguelkennedy

ramzxxx, 2019-04-10:
for anyone who has problems with spike app and your using "miao miao"
device with the libre cgm the "tomato" app from official apple app store
seems to work with loop app (V1.9.5) tomato sends miao miamo reading to
loop every 5 mins (in loop i have set the cgm setting to the insulin pump
(my pump minimed 722)

GrantGam, 2019-04-10:
@ramzxxx is this facility only available on X2X series pumps? I'm trying with my 554 but no joy...