SpikeApp/Spike #211: Weekly/monthly graphs in History

CoreISP, 2019-09-17:
Hi,

It’d be really nice if you could view data for a week and/or a month in History. Right now you have to view it day by day in 24H mode.
A weekly graph (both total and scrollable in 12 or 24H segments) would be very nice, it makes it much easier to see the curvature and compare days, see peaks occurring at the same time of the day, etc.

Hope this idea will be considered. :)
Thank you.

dxclancy, 2019-09-17:
Spike connects to Nightscout which gives you extensive controls over viewing the data. It may be that the developers feel that functionality is accessible to you vi Nightscout.

Cheun1, 2019-09-17:
Tu peux faire cela avec nightguard sur 5 jours .

closed 2021-05-07 by liroyvh