SpikeApp/Spike #214: Status icons not displaying on iOS 13

t1dpanic, 2019-09-25:
After upgrading to iOS 13, the status icons at the top of the device (network, time, battery, etc) will initially display fine, but eventually stop appearing in Spike. I can't consistently cause the issue, however it will inevitably occur and once it does they will not reappear until Spike is closed from the app switcher and restarted.

Tried deleting and re-installing the app (manual mode, apple developer account) with no change in behavior.

Issue occurs on both iOS 13 and 13.1 on iPhone 7.

[image]

m-a-v, 2019-09-25:
I can confirm this bug. It's reproducible when you switch to landscape mode.

t1dpanic, 2019-09-25:
Great, I can reproduce it by switching to landscape as well. 

Workaround is to enable the "No Rotation" feature in the "..." menu. This prevents Spike from ever going into landscape mode.

BenRepak, 2019-09-25:
This reproduces for me in landscape mode as well. dev branch installed 9/24/19, ios 13.1, iphone x