SpikeApp/Spike #216: Spike re-signer issues

Mindstrial, 2019-10-10:
I have issues with Spike re-signer. I'm using MacOS Catalina, Xcode 11.0, iOS 13.1.1 and WatchOS 6.0.1. I'm going with the installation tutorial on your page and there are 2 problems:
1. There is no HealthKit compatibility in Spike app (I don't really care about this much, but I think it's good to let you know)
2. After configuring identifiers and group for app, selecting my personal free Apple Account I run spike-resigner.sh but it fails installation with error: 
`Can't find watch extension's mobileprovision file. Please make sure you performed all Xcode steps correctly. Aborting!`
For now I've installed app using cydia impactor but I would love to have an app on my watch - right now it's impossible to do.

m-a-v, 2019-10-10:
https://spike-app.com/install-spike-using-xcode

Search for the following text: "can’t find mobile provision error" in the troubleshooting section. This should solve the issue.

Currently, it is not clear why this error happens. Let me know if it fixes your issue.

Mindstrial, 2019-10-10:
Oh wow, I feel stupid for missing that, sorry. Yes, this fixed me issue, thanks a lot.

On the other hand, the HealthKit Capability is still missing.

m-a-v, 2019-10-10:
Do you use a free account?

"Free accounts will not have the iCloud section enabled, only HealthKit and App Groups."

If no try to repeat all steps. Probably you missed something.

Spikey72, 2019-10-21:
> Oh wow, I feel stupid for missing that, sorry. Yes, this fixed my issue, thanks a lot.
> 
> On the other hand, the HealthKit Capability is still missing.

Like you, I am having the same error, can you tell me how your's was resolved?
I'm Using MacOS Catalina with Xcode 11.1, iOS 13.1.3, WatchOS 6.0.1

Was it:
4. For some unknown reason, sometimes even selecting the same Apple account in both Terminal and Xcode results in the “can’t find mobile provision error”. If that’s the case you might try opening the Spike Re-Signer file with TextEdit, add a # at the begging of:
Line 366: if [ “$CODE_SIGN_IDENTITY” = “$MOBILEPROVISION_IDENTITY_NAME” ]; then
Line 423: fi
Now save the Re-Signer file in TextEdit and run it again in Terminal.

Or:
Building the App on a Device that already has Spike Installed

Thank you

Spikey72, 2019-10-22:
> Oh wow, I feel stupid for missing that, sorry. Yes, this fixed my issue, thanks a lot.
> 
> On the other hand, the HealthKit Capability is still missing.

I noticed that too, I went through the setup three times.
There is a +Capability option at the top were you can add Healthkit but unsure if this is the correct way way of doing it. Looks like Xcode has changed in v11.1?

Ariandr, 2019-11-08:
The same issue.
I cannot select HelthKit capability, even though everything else is correctly set.

[image]

Dissolver, 2019-12-18:
> Can't find watch extension's mobileprovision file. Please make sure you performed all Xcode steps correctly. Aborting!

Faced the same with paid developer account, used dev certificate and provisions for signing.
MacOS Mojave (10.14.6), Xcode 11.3

benfiscal, 2020-03-15:
I added the HealthKit capability by the instruction Spikey72 mentioned above. Then I had to comment out 
Line 366: if [ “$CODE_SIGN_IDENTITY” = “$MOBILEPROVISION_IDENTITY_NAME” ]; then
Line 423: fi
as per the instructions in the troubleshooting section of the install-spike-using-xcode guide.

That fixed it for me.
I'm running macOS Catalina 10.15.3 and Xcode 11.1

closed 2020-11-25 by m-a-v

m-a-v, 2021-10-31:
@florisi This was pemantently commented out in 8512fb0. 

https://github.com/SpikeApp/Spike/blob/8512fb0d1d0ac56a438df8c7c9e232f4625e0836/re-signer/spike_resigner.sh

commit 8512fb0d (2020-11-25): Fixed #216 Can’t Find Mobileprovision File Error