SpikeApp/Spike #218: Spike app killing battery

mcsllb, 2019-10-28:
I love this app and am really grateful to the developers for giving me such great insight into my diabetes without costly CGM. However I have noticed an issue recently whereby Spike kills my iPhone XS battery when running in the background and also causes the phone to become extremely hot. This is usually resolved by closing the app and reopening it and doesn’t happen all of the time, only periodically. Although it does seem to be happening more frequently of late. 
[image]
[image]

Normally I would expect Spike to be using 10-15% battery when running in the background but have seen both of the above in the past few days. 

Please help!

Cheun1, 2019-10-28:
Bonjour .
la solution qui a marché pour pleins de personnes .

Si vous voulez revenir à quelque chose de propre.Avec 13.1.2 ou 13.1.3 ( la
13.2 je sais pas vu que je ne ferais plus de mise à jours ios étant en
13.1.2)

0/ Faites vos 2 sauvegardes spike via maintenance.
1/ Supprimez spike de votre iphone et de icloud.
2/ Attendez quelques heures.
3/ Faites une nouvelle sauvegarde icloud.
4/ Réinitialisez votre iphone. (effacer contenu et réglages)
5/ Mettez ensuite votre nouvelle sauvegarde icloud .
6/ Attendez quelques heures .
7/ Résignez spike de nouveau comme vous le faites d'habitude
8/ Scannez votre qr code à partir du mail sur votre ordinateur via
maintenance restaurer.
9/ ouvrez le mail de spike avec vos données sur votre iphone et choisissez
ouvrir avec spike et hop......

Cheun1, 2019-10-28:
iPhone 11 pro iOS 13.1.2
Avant c’était l’enfer et maintenant avec ma solution c’est redevenu normal

RobCvdV, 2019-10-31:
Did that really help?
I have the same issue. With also running Loop, my iphone SE with ios 13.1.3 is drained within half a day.
Spike running only for loop offline , so no predicted curve. But with http server.

RobCvdV, 2019-11-15:
I have had an update to ios 13.2.2 now and it seems to be going much better now. No more hot phone and still at 81% at 16:52 (4:52 am).
I left my apple watch at home as well. So could still be that. Testing that tomorrow

seattlebrighton, 2020-09-26:
@RobCvdV RobCvdv I'm still seeing this battery drain issue with loop and soiled and iphone 8 iOS 13.7. Did you decide if the improvement was due to new ios or to apple watch not connected?

closed 2024-05-12 by m-a-v