SpikeApp/Spike #221: Dexcom G6 - 8H - NO connection 2 Spike

franklance89, 2019-11-19:
8G**** are said not to be compatible, I would assume H will not be either
any letter above G - our next transmitter will be I (eye), and we anticipate it not working
As you can read in xdrip group , xgroup was recoded to work with the new transmitter.
Spike developers have to implement the same in Spike to work with the new transmitter
---------------

Hi guys, I am experiencing problems restarting a new trasmitter of Dexcom G6. I’ve read two SOLVED posts about it, but nothing seems to work. I get the paring request and this is it.

I initially started on saturday morning, and by lunch time decided to install the official app to get me though the weekend.

Today I deleted the Dexcom app and the transmitter among my bluetooth device list and give it a serious try with Spile again.
I have tried soft reset, stop and starting the sensor again, with and without skipping warm out, closing the app counting to ten and reopen it, locking the screen, chaging trasmitter to Miaomiao reopening and insering the G6 code again, deleting the transmitter from my bluetooth device list and starting again, also restore my clean back-up... but nothing 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

alwayssimifuzi, 2019-11-29:
Hey did someone have an solution for this? I have the exact same problem. There is also no initial-calibration popup, tried  very agressive mode + alt. Version 1, 2 and 1-2

spicypeppers, 2020-06-09:
Installed Loop/Spike... great. Followed above.
Another 8Hxxxx issue.
Is this solved?

mwood34, 2020-08-08:
I have just moved from the G5 to the G6 and have the same problem, transmitter is 8LXXXX. Does anyone know if there is a solution?

MWILSON476, 2020-11-13:
This still hasn't been fixed. I spent about five hours in the entire installation and trouble shooting before going to the Spike FB and reading that all of the newer transmitters from Dexcom are unable to be read.

Mad!