SpikeApp/Spike #222: Assistance when installing

Billspikedexcon6, 2019-12-15:
while installing the Spike app i get to the point in the install where you enter your id and password for the Cydia Impactor. 
This is the message displayed: file: installer.cpp; line 71, what: _assert(!teams.empty0)
![image](https://user-images.githubusercontent.com/58910876/70864629-dff55300-1f21-11ea-9228-7997fbd9c0c9.png

zshh, 2019-12-19:
Had the same issue. Apparently Cydia Impactor is broken at the moment. Read somewhere that it will take few weeks until developer has sorted it out.

Since i have mac, i went for the other option of installing with xcode and that worked without any issues

Dune-jr, 2019-12-28:
I echo this. Same issue, fortunately the manual [Install Spike using XCode](https://spike-app.com/install-spike-using-xcode/) still works well. If you struggle with that, come hang around https://gitter.im/SpikeiOS/Lobby for some help, @Billspikedexcon6

simonRos, 2022-06-04:
Bumping. Still an issue

LiroyvH, 2023-03-04:
@simonRos 
> Bumping. Still an issue

And it's extremely unlikely it'll ever be fixed. Cydia Impactor is dead. Saurik isn't maintaining it anymore. It's broken and will stay broken, so forget installing with Impactor. 

If you wish to install Spike, you'll require XCode or leverage AltServer. Installation instructions for both can be found in our Facebook group as mentioned on the website.