SpikeApp/Spike #223: MiaoMiao2 support?

iSelant, 2019-12-17:
I somehow managed to pair my MM2 with Spike but it won't read the glucose data from it. Is this new transmitter going to be supported?

Cheun1, 2019-12-17:
Le mm2 marche trés bien avec spike sauf si tu as mis à jour son firmware de
4 vers 7 avec l'application Tomato , si c'est le cas la solution est sur le
groupe Spike.

iSelant, 2019-12-17:
Oups c'est exactement ce que j'ai fait.. tu aurais un lien vers le groupe? merci de ta réponse rapide

Cheun1, 2019-12-17:
Voilà le lien du groupe la solution est dans annonce du groupe et le
fichier pour downgrader c'est moi qui l'ai.

https://www.facebook.com/groups/spikeapp/?ref=group_header

iSelant, 2019-12-17:
J'ai envoyé une demande pour rejoindre le groupe. Je peux trouver le fichier dans annonce du groupe du coup ou il faut que tu me l'envoie ? (si ca ne te dérange pas bien sur)

Cheun1, 2019-12-18:
Bonjour il faudra que que je te l'envoi

iSelant, 2019-12-18:
Bonjour d'accord. Mon fb c'est Antoine Prn dans le groupe, j'attends le fichier avec impatience :) merci

closed 2019-12-18 by iSelant

MazZilaaaa, 2020-01-28:
What about mm2 and libre 2

Cheun1, 2020-01-28:
Hello MM2 with libre 2 (not us 14) days works with spike

m-a-v, 2020-01-28:
@MazZilaaaa Currently not supported by Spike.

randallward-dk, 2020-01-28:
Hej Sascha,

Its working for me with an libre 1 / 14 day sensor/iPhone Xavier. There have been problems with Tomato app upgrading the firmware, then it doesn’t work.

Randall

Sendt fra min iPhone