SpikeApp/Spike #229: I am assuming this project is dead.

meluvalli, 2020-04-24:
I am assuming this project is dead as it hasn't been updated in almost a year and no longer works with directions provided?

m-a-v, 2020-04-24:
The project is not dead. If you need up-to-date information, please visit the Facebook group.

closed 2020-04-24 by m-a-v

meluvalli, 2020-04-24:
Hmm.  Ok.  Just assumed because the directions are out-dated and won't compile in XCode 11.   Don't have Facebook account, so guess out of luck.

wireman642, 2020-04-24:
If you read the build info, parts of the app aren’t included in the github because they are closed source. They can’t include it, so the source on github is incomplete. 

From my understanding, it would need to be rebuild from the (private) source in the modern Xcode. Likely won’t happen unless there is a good reason.