SpikeApp/Spike #237: Error" Can't find watch extension's mobileprovision file. Please make sure you performed all Xcode steps correctly. Aborting!"

Steel121, 2020-10-23:
When I run the resigning script I get this error. I DID comment out lines 366 and 423 but still receive this error.

Is there something else that I need to cmment out?

m-a-v, 2020-10-23:
Not that I know of. Did you follow all the other steps correctly?

Steel121, 2020-10-23:
I think so but the guide also uses a different version of xcode with a developer account so the screens don't match  1 for 1. I am using a free account.

m-a-v, 2020-10-23:
The manual on the homepage is no longer up-to-date. I will transfer it to GitHub and update it. In the meantime, it's best to use the instructions in the Facebook group. Thanks for your understanding.

Steel121, 2020-10-23:
I don't have an active facebook account. Any chance you can paste?

m-a-v, 2020-10-23:
Sorry, this is not possible. I recommend you to create an account. Even if you need more information and help. Or you wait until I have updated the installation manual which will certainly take some time.

Steel121, 2020-10-23:
ok thanks for the quick responses

closed 2020-10-23 by m-a-v

prewk, 2020-10-26:
@m-a-v Please write a comment in this issue when you've updated <3