SpikeApp/Spike #238: Support for "Libre Pro" Sensors

anirudhreddy28, 2021-03-19:
We notice that Spike currently does not support Libre pro sensors. We in India primarily use Libre Pro as our CGM and we are struck with using the Native apps like Tomato (for Miao Miao) and Link Blucon (Ambrosia Blucon). Most of them are happy with the Native apps but with the recent server issues Miao Miao had, we are interested to see a possibility of using these DIY apps. There are hundreds (possibly thousands) who could benefit with this addition.

We know that Spike already supports Libre 1, Libre US 14 day, Libre US 10 day etc.. We would like the similar support for Libre Pro as well

At this moment only alternatives we have have are the native apps and they have been unreliable in many times and also providing us limited functionality for those who are interested in DIY Closed loop technologies.

I am personally on DIY Closed loop using Libre Pro + Tomato + Nightscout. There are few others using exactly the same setup. But with this setup, we can use it only when we are connected to internet. If we could get the Spike support Libre Pro sensors, it would be helpful to do closed loop in offline mode as well.

jain-shreyansh, 2023-08-05:
Any progress in this?

m-a-v, 2023-08-05:
There is currently no active development. So the answer is no.

Also [xDrip4iO5](https://xdrip4ios.readthedocs.io/en/latest/) does not seem to support the Libre Pro.

closed 2023-08-05 by m-a-v

gui-dos, 2023-08-06:
@MarekM60 ( R.I.P. :-( ) passed me some internal info and even a firmware dump of the Libre Pro (https://github.com/gui-dos/DiaBLE/blob/main/Reversing/Dumps/LibrePro-dump.bin) and I started writing a class to support its special FRAM and extended header layouts in iOS (https://github.com/gui-dos/DiaBLE/blob/main/DiaBLE/LibrePro.swift).

I was reported by some Indian guys that they could also connect to a Libre Pro successfully via a MiaoMiao but I never tested real samples and programmed in Java...