SpikeApp/Spike #240: IOS 14 update

Millercio, 2021-06-03:
Today the new version of IOS has come out, the 14.
After installing the update on the phone I tried to open Spike but I got an error message, which said:
"The developer must update this App to work with IOS 14"

Does anyone know if there is already a Spike update, and how it has to be installed?

Thank you very much to all

liroyvh, 2021-08-05:
You simply need to re-sign with latest software (AS/XCode/Impactor) to fix it.

closed 2021-08-13 by m-a-v

simgislab, 2021-09-07:
iOS 14.7.1
Cydia Impactor
Current version of the app from 
https://spike-app.com/releases/latest-iphone-impactor-version

If you're having:
[image]

Solution: update Cydia Impactor to latest version