SpikeApp/Spike #243: Spike does not update, despite « very aggressive »…

BaQs, 2021-09-21:
Since I installed Spike, it does not follow up new values. It stays « frozen » at the time the app was launched.
Treatments can be updated.
But graph stays still. I tried every single mode: from std to very aggressive, and the 2 alternatives also.

I see also that predictions are never working. It says « N/A ».

m-a-v, 2021-09-21:
I assume you are using xDrip4iO5 as master? See the following issue here:

See here: https://github.com/JohanDegraeve/xdripswift/issues/262

BaQs, 2021-09-21:
I do!
hmm, crazy issue. Let me +1 then

BaQs, 2021-09-21:
> I assume you are using xDrip4iO5 as master? See the following issue here:
> 
> 
> 
> See here: https://github.com/JohanDegraeve/xdripswift/issues/262

So by using xdrip calibration I do have now the predictions. Thanks.

But my basal is still shifted
Any idea ?

🙏