SpikeApp/Spike #246: Can I use the same Apple ID to install spike in different iPhones?

HelomSilva, 2022-02-06:
I tried to install spike in 3 different iPhones (one to my son T1 and the others to me and my wife follow him). In the first one, the app works and in the others not. In the iPhones that is not working, the app is closing. What happened? Have you any idea?

m-a-v, 2022-02-06:
Yes. You can install it using the same Apple ID as long as you’ve registered the phone UUIDs on your Apple developer account.

if you need more help I would suggest that you join the Facebook group.

closed 2022-02-06 by m-a-v

HelomSilva, 2022-02-06:
> Yes. You can install it using the same Apple ID as long as you’ve registered the phone UUIDs on your Apple developer account.
> 
> 
> 
> if you need more help I would suggest that you join the Facebook group.

All 3 devices are registered on my developer account but only one works.