SpikeApp/Spike #248: Spike does not update

HelomSilva, 2022-02-07:
Since I installed Spike, it does not follow up values. It stays « frozen » at the time the app was launched.

I am using Miao Miao 3 as transmissor. Any idea how to solve this problem?

iOS 15.2.1
Iphone 12 Pro Max

tannygamer007, 2022-02-12:
I too am facing Same issue.in iPhone 6 with miao miao 3.How did you push the spike code to iPhone 12 pro Max?? I was unable to push it to iPhone xs Max and iPhone 12.

tannygamer007, 2022-02-12:
And what spike version are you currently running on iPhone 12 pro Max???you can check it in settings of spike and about section..

jawngee, 2022-02-16:
Are you using a US Freestyle Libre 2?

LiroyvH, 2023-03-04:
@tannygamer007 Late reply is late, but who knows - maybe someone has the same questions and finds this. So: install instructions for latest iOS version are only in the Facebook group. However, with a MM3 you can spare yourself the trouble - we don't support that transmitter in Spike. However, we do support it in xDrip4iOS. Note that for FSL2 we only support the EU model; which does not require a transmitter-bridge at all, direct connect.

closed 2023-03-04 by m-a-v