SpikeApp/Spike #250: spike app won't resigned

HDBAPro, 2024-05-10:
Spike app won't resign
```
Selected code signing identity -> Apple Development: XXXXXX
-e Detected account type -> Paid Developer Account
-e Spike version -> 3.6.3

spike_resigner.sh: line 825: syntax error near unexpected token `<'
spike_resigner.sh: line 825: `done < <(find "$TEMP_DIR/Payload/$APP_NAME" -d -mindepth 1 \( -name "*.app" -or -name "*.appex" \) -print0)'
```

m-a-v, 2024-05-12:
If you need help please join the Facebook group. Other instructions are no longer up to date and there will most likely be no support here.

closed 2024-05-12 by m-a-v

HDBAPro, 2024-05-18:
I had installed Spike app before but wenn I try to extend resign It gives me this error
```
$ sh spike_resigner.sh

███████╗██████╗ ██╗██╗  ██╗███████╗    ██████╗ ███████╗    ███████╗██╗ ██████╗ ███╗   ██╗███████╗██████╗
██╔════╝██╔══██╗██║██║ ██╔╝██╔════╝    ██╔══██╗██╔════╝    ██╔════╝██║██╔════╝ ████╗  ██║██╔════╝██╔══██╗
███████╗██████╔╝██║█████╔╝ █████╗      ██████╔╝█████╗█████╗███████╗██║██║  ███╗██╔██╗ ██║█████╗  ██████╔╝
╚════██║██╔═══╝ ██║██╔═██╗ ██╔══╝      ██╔══██╗██╔══╝╚════╝╚════██║██║██║   ██║██║╚██╗██║██╔══╝  ██╔══██╗
███████║██║     ██║██║  ██╗███████╗    ██║  ██║███████╗    ███████║██║╚██████╔╝██║ ╚████║███████╗██║  ██║
╚══════╝╚═╝     ╚═╝╚═╝  ╚═╝╚══════╝    ╚═╝  ╚═╝╚══════╝    ╚══════╝╚═╝ ╚═════╝ ╚═╝  ╚═══╝╚══════╝╚═╝  ╚═╝
v1.0

-e
Do you want me to clear all previous Spike certificates from your Xcode cache? This is needed if you want to renew the expiration date of your currently installed Spike. If you're signing Spike for the first time or just updating to a new Spike version using a paid Apple Developer Account you can choose "NO".
1) YES
2) NO
3) CANCEL
#? 2
-e
Do you want me to download the lastest available Spike version?
1) YES
2) NO, I've already added my own Spike IPA file to the folder
3) CANCEL
#? 2
-e
Please select an IPA file:
1) Spike-iPhone-iPodTouch.ipa
2) Spike.iPhone-3.6.3.ipa
3) Spike.iPhone-iPodTouch.3.6.3-Resigned.ipa
4) CANCEL
#? 2
-e
Selected IPA file -> Spike.iPhone-3.6.3.ipa
-e
Preparing file structure...
-e
Please select a code signing identity:
1) Apple Development: XXXXXXX
2) CANCEL
#? 1
-e
Selected code signing identity -> Apple Development: XXXXXX (XXXXXXX)
-e Detected account type -> Paid Developer Account
-e Spike version -> 3.6.3

spike_resigner.sh: line 825: syntax error near unexpected token `<'
spike_resigner.sh: line 825: `done < <(find "$TEMP_DIR/Payload/$APP_NAME" -d -mindepth 1 \( -name "*.app" -or -name "*.appex" \) -print0)'
```