SpikeApp/Spike #252: Can't resign

r0ssr0ss, 2024-06-10:
I've followed the tutorial to the letter, I've done this yearly for the past couple years without issue. This time every time I try, right after I select my code signing identity I get this error?

Temp/Payload/Spike.app/PlugIns/fswidget.appex: replacing existing signature
Temp/Payload/Spike.app/PlugIns/fswidget.appex: resource fork, Finder information, or similar detritus not allowed
Encountered an error, aborting!

I have no idea what this means? I've googled it and can't find a solution? Any help?

m-a-v, 2024-06-10:
Did you try this?

https://stackoverflow.com/questions/39652867/code-sign-error-in-macos-monterey-xcode-resource-fork-finder-information-or

r0ssr0ss, 2024-06-11:
I did try, not sure what I'm doing. I"m not really that tech savvy, just looking for a solution. I ran all the prompts it said but still getting the same error returned back.

closed 2024-10-18 by m-a-v