SpikeApp/Spike #253: Error installing and signing on Mac OS X Sequoia - IOS 18

dpecile, 2024-10-10:
Hi
Apparently the location of the certificates, changed, from 
"${HOME}/Library/MobileDevice/Provisioning Profiles"
to 
"${HOME}/Library/Developer/Xcode/UserData/Provisioning Profiles"

I changed the signing script, but, after succefully signed the IPA, I have the error: 

0xe8008016 (The executable was signed with invalid entitlements.)

Any idea ? 
I am using the free account.

Thanks ins advance.

m-a-v, 2024-10-19:
As you've mentioned there is a different handling regarding path for Xcode 16. I've created a PR. Feel free to test it with this version. I was only testing it with the paid account and it works flawlessly on my side.

m-a-v, 2024-11-08:
Fix merged to main. Please test and report back. I will close the issue for now.

closed 2024-11-08 by m-a-v