SpikeApp/Spike #254: TODO: Add Sequoia compatibility for spike_resigner.sh

LiroyvH, 2024-10-17:
In MacOS 15 (Sequoia), the place for storing the provisioning profiles has moved. 

This means that this line:
`MOBILEDEVICE_PROVISIONING_PROFILES_FOLDER="${HOME}/Library/MobileDevice/Provisioning Profiles"`
Has to read:
`MOBILEDEVICE_PROVISIONING_PROFILES_FOLDER="${HOME}/Library/Developer/Xcode/UserData/Provisioning Profiles"`

However, considering we also still have users on older MacOS versions: ideally we add a check to look for path 1 on MacOS < 15 and for path 2 on MacOS > 14. I have not yet had the time to write such a check. If someone wants to: please PR. Otherwise, please be patient.

Interim, a modified version of spike_resigner.sh that works on MacOS 15 **and only on MacOS 15** can be downloaded from https://liroy.pro/spike_sequoia.sh. The usage is the same as spike_resigner.sh, just you launch "bash spike_sequoia.sh" instead. (Note: depending on permissions, you might have to use "sudo bash spike_sequoia.sh". If you get a f-ton of "Permission denied" errors scrolling over your screen when you get to the signing step: fix the folder permissions or use sudo.)

m-a-v, 2024-10-19:
@LiroyvH  Please verify https://github.com/SpikeApp/Spike/pull/255.

m-a-v, 2024-10-20:
Important: This is not related to the macOS version, but rather the Xcode version which is reflected in the adapted script. Tested on macOS Sonoma and macOS Sequoia (still a little bit wondering why they name two subsequent macOS versions starting with the letter S ...).

closed 2024-11-08 by m-a-v